idolpx/mobile-rr #1: console - "beep rr" command doesn't work

idolpx, 2016-07-09:
When you enter "beep rr" in the console it should play the startup tune.
It doesn't though.  I'm not sure why.  The same function works fine at bootup.

boneskull, 2016-07-29:
likely related to esp8266/Arduino#836

idolpx, 2016-07-31:
Thanks for the info and for posting about it on that thread.  I hope a solution is found.  It really doesn't make much sense to me right now.  I'm still trying stuff to try to get it working.

boneskull, 2016-08-01:
I messed with it for a bit and could only assert the code was running. almost positive it's not a bug in your code rhoigh

boneskull, 2016-08-01:
Though

idolpx, 2016-08-03:
Thanks for looking into it.  I was hoping an update to the SDK might fix it.  It's not a critical function and it's nice that it works on boot up at least.  :)

Hopperpop, 2016-08-17:
I think that you can't use delay() or yield() in an event. If you call beep_rr() in loop(), it works perfectly.  I tested this with this code:
```C++
enum class statemachine{
	none,
	beep,
	rr
};
statemachine state = statemachine::none;
int beepv;

void loop ( void )
{
	dnsd.processNextRequest();
	ArduinoOTA.handle();  // Handle remote Wifi Updates

	switch (state){
		case statemachine::rr:
			beep_rr();
			state = statemachine::none;
			break;
		case statemachine::beep:
			beepC(beepv);
			state = statemachine::none;
			break;
	}
}
```
Instead of calling the beep functions in the web-socket event, change the state variable to the desired state. Note that this will block the dns proces for a short time when playing a note/song.

idolpx, 2016-08-21:
Oh awesome!!!  I'll check it out.  Thanks for the feedback!

idolpx, 2016-08-21:
Works like a charm.  :)  Thanks again.

closed 2016-08-21 by idolpx

boneskull, 2016-08-22:
@Hopperpop Nice catch & solution; didn't know that about the event handler.