idolpx/mobile-rr #2: Android Captive Portal Helper isn't triggered.

idolpx, 2016-07-09:
When connecting on an iOS or OSX device the captive portal is detected and a browser window pops up like it should to be able to accept the agreement and then get rick roll'd.  It doesn't launch the browser window on Android though.

ratmandu, 2016-08-12:
Just tested with Android 6.0.1 on a Nexus 5X, and a Nexus 6. Connecting to the hotspot brings up a "Sign in to wifi network" notification and toast. Clicking the notification loads the captive portal.

What version of android are you running?

idolpx, 2016-08-15:
I'm using a Galaxy S5 with Android 6.0.1.  I'm not sure why my phone doesn't work the same.  I'd like for it to open a browser page like it does on iOS.  It does pop open a browser when I connect to some captive portals at restaurants and parks with free wifi though.  I'd like for mobile-rr to work the same as those captive portals.  I just need to do some packet sniffing to find out what the difference is.

Pukinator, 2016-08-19:
The captive portal also doesn't work for me although I'm on iOS. Is it possible that it's broken on iOS 10?

idolpx, 2016-08-20:
I don't have an iOS 10 device but will see what I can come up with to test with.

Pukinator, 2016-08-20:
It just worked once for iOS 10, while it works again if I renew my release in iOS 9

Noki, 2016-08-20:
Please check if my pull request fixes it: https://github.com/idolpx/mobile-rr/pull/10

idolpx, 2016-08-21:
Hmmm... still the same for me on my Android device but if it fixed it for you I'll merge it.  
Thanks.

Pukinator, 2016-08-21:
This doesn't fix it for iOS 10

Noki, 2016-08-21:
Hmm. Stopped working for me as well. Strange thing is that it sometimes works. It might be a caching issue. Probalby it is a good idea to send nocaching Headers for the redirects and the captive portal index and set the TTL of DNS answers to 0 seconds.

The DNS setting can be set using: 

`dnsd.setTTL(0);`

And the caching directives could be set using something like this:

```
AsyncWebServerResponse *response = request->beginResponse(302);
response->addHeader("Cache-Control","no-cache");
response->addHeader("Pragma","no-cache");
response->addHeader ("Location", "http://10.10.10.1/" );
request->send(response);
```

However that still was not enough. But maybe because stuff is still cached and I don't know how to reset the DNS-Cache on my phone and other caches.

idolpx, 2016-08-27:
Thanks for the suggestions.  I'm adding them now.  I still have to do some WiFi sniffing to try and determine the difference between this captive portal and others.  

Is there a way to set DHCP option 160.  I'm curious if that is what the difference is.

Starbix, 2016-08-27:
It looks like it works on iOS 10 now.

idolpx, 2016-08-27:
Oh great!  Thanks for the feedback.

tpmodding, 2016-10-11:
on my ios 10.0.1 iphone 6 doestn work anymore...i use a nodemcu and had to change the platformio.ini to this:

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
build_flags = -ULWIP_OPEN_SRC -Wl,-Tesp8266.flash.4m.ld

than compiling and uploading is no problem anymore...but as said the portal doesnt open...can only go to console if i type the url :/

treii28, 2017-09-19:
Not working for me either, but I can't imagine it being a device dns-cache issue because places like McDonald's still cause a pop-up on my phone. Has anyone tried sniffing the wifi when connecting to a mcdoogles hotspot? Tell me a good program to grab wifi frames on connection and I'll try it myself.

treii28, 2017-09-23:
Further testing on my part with your dns server library, and I can see the dns requests are being answered, but I never see the requests combing back to the web server. My guess appears to be dns caching or otherwise not-trusting the response of the local-net address on the part of the mobile device.  I'm wondering if there is any way to get the web server code to answer the /generate_204 and other urls regardless of the address they are being sent to. Something similar to IPTables redirects in linux or something.
I see they have an lwip implementation in the esp8266 code and some of the libraries (including ESPAsyncTCP) use it for some of their protocol handling. Maybe there's some way in there (it's a tad beyond my skill level at the moment, but I'm trying to find resources to learn as much as I can about it) that could help?

idolpx, 2018-01-05:
I added a callback to be able to override the IP of a DNS query to point the DNS to an IP that is inactive on the local network and that did not solve the issue either.

This post suggested that method.
https://socifi-doc.atlassian.net/wiki/spaces/SC/pages/94371841/DNS+Workaround+to+keep+Android+Splash+Page+and+the+Captive+Portal+Notification+active

I'm just now getting back to this to see if I can come up with a solution.  My phone pops up the CP Helper every time at a McDonalds or Taco Bell and other CPs at restaurants but not from this little device.  I will do some sniffing again soon to see if I can find any differences.

Any feedback is greatly appreciated.  I'd love to solve this issue.

drzony, 2018-02-08:
I'm working on a bit different project which uses Captive Portal. I have rewritten DNS code from scratch using AsyncUDP, extended it to handle most of DNS requests and it still did not help (manual browser navigation redirects correctly).
The problem seems to be limited to Samsung devices. I have S7 edge and have no luck with pop-up (only "Internet may not be available" shows). On my Tab Pro with Cyanogen installed it works as intended.
It seems that my phone does not request /generate204 from web server (nor does it make any other requests). I've looked into AOSP code and it should visit those URLs, so it seems that Samsung has changed the way that CP detection works.
Still I was not able to determine what triggers the pop-up on my phone. Maybe there is some IP whitelist and/or some special DNS answer required.

idolpx, 2018-02-17:
Did you try to implement DNS Option 160 in your customer DNS code?  I'm curious if that would activate the CP Helper.

drzony, 2018-02-17:
There is no such thing as DNS option 160, it's DHCP option 160 and it's implemented in DHCP server on ESP (I have verified that), if it was not implemented then the device would not receive address of DNS server (and it would not receive any DNS query packets). From inspection of packets coming to my DNS server I can see that the phone asks for the IPs for google servers and my ESP sends responses (I've verified that those responses are valid). But there are no HTTP requests made to web server, which is weird. If I open Chrome manually then if I type http://github.com then I'm properly redirected to my captive portal (which means that both DNS and web server are working), but there is no "Sign in to WiFi network" popup. As I've written in my post before on my Tab Pro with Cyanogen everything works as expected.

idolpx, 2018-02-18:
Ooops sorry, I read your message too fast... Yes it is for DHCP but it is not for sending the DNS server address.  It is for sending the url for a captive portal within the DHCP configuration.

This article explains it's use a bit better.

http://community.arubanetworks.com/t5/Technology-Blog/RFC-7710-Captive-Portal-Identification-Using-DHCP-or-Router/ba-p/255737

drzony, 2018-02-18:
Unfortunately this would require change in ESP SDK, you can see DHCP options here https://github.com/esp8266/Arduino/blob/f9ac524b13348e18a1ceb00261d947d6c1e0f9b5/tools/sdk/lwip/src/app/dhcpserver.c#L134
Captive portal is only a small part of my project, so checking this out will take some time (and I wonder if it's worth the effort)

idolpx, 2018-05-10:
Oh shoot... ok.  Thanks for trying it out and letting us know.
This bugs me every time I think about it.  It has been a while.
Getting this working consistently across all devices would be the icing on the cake for the project.

vidharv, 2018-08-16:
Did you guys manage to get it working!??

idolpx, 2018-08-25:
I have not looked at it again in a long time.  Still an open issue until we find a solution.

Marfjeh, 2018-10-09:
I got it working!

there are some commeted out lines on `mobile-rr.ino` line 576 ish..

remove the comments and adding

```cplusplus
                if ( strstr(domain, "connectivitycheck.gstatic.com") )
                    dnsd.overrideIP =  IPAddress(74, 125, 21, 113);
                // connectivitycheck.android.com -> 74.125.21.113
                if ( strstr(domain, "connectivitycheck.android.com") )
                    dnsd.overrideIP =  IPAddress(74, 125, 21, 113);

                // dns.msftncsi.com -> 131.107.255.255
                if ( strstr(domain, "msftncsi.com") )
                    dnsd.overrideIP =  IPAddress(131, 107, 255, 255);
```
It works! it seems my phone (s7 edge) was sending the request to `connectivitycheck.gstatic.com` not to `connectivitycheck.android.com`

I also enabled `dnsd.setTTL(0);`

Noki, 2018-10-10:
@Marfjeh Good job! Do you send in a pull request for all the lazy people? ;-)

Marfjeh, 2018-10-10:
Sure. i'll fork the project

Marfjeh, 2018-10-10:
made the pull request. #30

idolpx, 2018-10-18:
Awesome!  I will check this out as soon as a get a moment.

idolpx, 2018-10-18:
I was looking at my code that I hadn't committed and at some point I did try "connectivity.gstatic.com" but it didn't seem to work on my Galaxy S5.  This is what I had to cover all of the android check domains.

```
// connectivitycheck.android.com -> 74.125.21.113
if ( strstr( "clients1.google.com|clients2.google.com|clients3.google.com|clients4.google.com|connectivitycheck.android.com|connectivitycheck.gstatic.com", domain ) )
    dnsd.overrideIP =  IPAddress(74, 125, 21, 113);

```

Going to test this again shortly with on my S5.

Looks like a few of the comment lines in your pull request were uncommented.

idolpx, 2018-10-18:
It's good to hear it is working on the S7 but unfortunately I still get the same results on my S5.

Marfjeh, 2018-10-19:
Wierd, which android version are you using?

idolpx, 2018-10-22:
It's Android 6.0.1. The latest update from Verizon.

If I go to a place with free wifi and connect I get the popup so I know it is capable.  I just don't know what is different about how I'm handling it in this code versus what is going on with the captive portal at restaurants and other places where it works.

Marfjeh, 2018-10-22:
hmm. i wonder what packets it sends... maybe we capture it with Wireshark?

idolpx, 2018-10-22:
Yeah... I did some captures to try and compare at one point but didn't find any clues.  I have been meaning to get back to it at some point.  It would be nice to solve this mystery.

Noki, 2018-10-22:
according to some pages I found it's "connectivitycheck.gstatic.com" for Android 6:

https://enterprisenetworkingatlarge.wordpress.com/2018/04/21/captive-portal-detection-on-android-and-others-client-vendors-vs-ap-vendors/

drzony, 2018-10-22:
According to my research Samsungs with Android 6 and 7 do something differently. When checking the logs from ESP DNS I have noticed that they ask for much more pages that connectivitycheck. I think they may be related to some "social services" login like Facebook or something like that. I have updated my S7 to Android 8.0, but haven't had the time to verify if it changed anything. The easiest way to verify what DNS requests are made is to add some logging on ESP. However I have tried to redirect all of them to captive portal using DNS and then combinations of HTTP redirects and it failed. Samsung has made some additional "checks" that are required to show "sign in" popup.

Noki, 2018-10-22:
ok, so maybe we have to add a form to the html and hide it with CSS ;-)

drzony, 2018-10-22:
Does not help, I have a captive portal with a proper form on my project and it still does not work. Maybe Samsung added some MAC filter for Access Points or some weird "smart" checks. As I wrote in a post before, going manually to some page in Chrome works like a charm.

Marfjeh, 2018-10-22:
then they need to add MAC addresses on the whole world, thats not going to work. but i can agree samsung maybe expects something

Noki, 2018-10-22:
There is an RFC for Captive-Portal Identification Using DHCP or Router Advertisements: https://tools.ietf.org/html/rfc7710

I think this would be the correct approach. However it works totally different than the method currently implemented.

Marfjeh, 2018-10-22:
Hmm intresting!

drzony, 2018-10-22:
RFC7710 details DHCP option 160 which was tested above in https://github.com/idolpx/mobile-rr/issues/2#issuecomment-382128739

idolpx, 2018-10-22:
I was able to find the source code for the firmware on my phone here.

http://opensource.samsung.com/reception/receptionSub.do?method=sub&sub=F&searchValue=G900V

Maybe there will be a clue in there.  I'll take a look and see what I can find.

idolpx, 2018-10-23:
No luck with that code.  :(

idolpx, 2018-10-24:
Found this post with others having this same issue.  Their solution describes what we were just doing. 

https://unix.stackexchange.com/questions/432190/why-isnt-androids-captive-portal-detection-triggering-a-browser-window

I need someone else with a Galaxy S5 or other device that is having this same issue to confirm and compare.  A device test chart might be useful for figuring this out.

I was just wondering if changing the device default IP to 192.168.1.1 would make a difference.  The current setup is a Class A private subnet but I also have it using a Class C subnet mask.  That should be corrected.  Going to test that next.

idolpx, 2018-12-05:
It looks like we are missing a port forward in LwIP to localhost for any IP request to port 80.
I think the following project may be able to be used to get that done via it NAPT implementation.

https://github.com/martin-ger/esp-open-lwip

I'm looking into it now but not sure how it would be easily implemented in this project.
Any ideas or help are greatly appreciated.

drzony, 2019-02-25:
I finally managed to make it work, it turns out that DNS redirects for:

* clients3.google.com
* clients.l.google.com
* connectivitycheck.android.com
* connectivitycheck.gstatic.com
* play.googleapis.com

need to be answered with invalid address e.g. 192.168.4.11 (if softAPIP is 192.168.4.1)
I've just verified that this works, but it needs custom DNS server, because the one that is included in ESP Arduino lib always answers with single IP.

idolpx, 2019-02-25:
I put in a callback so you can override the IP address being returned from DNS.  But from what I was finding it has to be a non-local IP address and you also have to override all connections to port 80 on any IP and direct them to the softAPIP to get it to work correctly. 

I will try this though and see if it works for me too.  :)  Thanks for the feedback.
This one has been bugging me for a long time.

drzony, 2019-02-26:
You do not need IP redirection. Here is my exact setup that works:

1. DNS server answering with:

```
IPAddress dest_address = WiFi.softAPIP();

if (strcmp(label, "clients3.google.com") == 0 || strcmp(label, "clients.l.google.com") == 0 ||
    strcmp(label, "connectivitycheck.android.com") == 0 ||
    strcmp(label, "connectivitycheck.gstatic.com") == 0 ||
    strcmp(label, "play.googleapis.com") == 0) {
    dest_address[3] += 10;
}
```

2. HTTP server redirecting "notFound" and everything that is not meant for "myhost.manager.net" to "http://myhost.manager.net", with the following redirect:
```
if (request->host() != "myhost.manager.net" && request->host() != WiFi.softAPIP().toString()) {
    AsyncWebServerResponse *response = request->beginResponse(307);
    response->addHeader("X-Frame-Options", "deny");
    response->addHeader("Cache-Control", "no-cache");
    response->addHeader("Pragma", "no-cache");
    response->addHeader("Location", "http://myhost.manager.net");
    request->send(response);
}
```

And that's all that is needed.

treii28, 2019-03-06:
Let me know as soon as there is something final to try out. I would love to get that project down off the 'waiting for resolution' shelf.

treii28, 2019-03-06:
@drzony is your dnsserver code posted anywhere?

drzony, 2019-03-06:
Not yet, it's part of a bigger project and it will take a while to release, you can modify included DNSServer in line https://github.com/idolpx/mobile-rr/blob/41841a67682127afd9d283569e4a7065b8143178/lib/DNSServer/DNSServer.cpp#L59
Just add comparisons with addresses from my comment and then add replyWithFakeIp()

treii28, 2019-03-07:
but isn't _domainName in that instance the one specified when the server instance is created? It appears to me the UDP _buffer is what is going to contain the incoming request but it's going to be in the raw UDP packet format (with the domain name started at I believe position 13 [14th digit])

treii28, 2019-03-07:
btw, if you control your own code and always plan to use low 4th digit ips, you should be fine, but I'd probably do something like this:

if(dest_address[3] < 244)
            dest_address[3] += 10;
else
            dest_address[3] -= 10;

treii28, 2019-03-07:
untested - I'll try it out later this evening:
[https://github.com/treii28/DnsServer](modified DnsServer for captive portal)

drzony, 2019-03-08:
@treii28 concerning dest_address[3] += 10, it does not matter since it's uint8_t, so it will overflow to zero automatically. In terms of address clashes it should also work as the address I'm returning needs to be fake, any connection to broadcast will fail, so it should trigger the sign-in popup.
As for you code, if you are using DNSServer from this repo you can provide callback to onOverride and change the address there (sorry for the confusion, I did not notice earlier that the callback option was already added)

treii28, 2019-03-08:
ok, wasn't sure on the type - but overflow to zero could also equal zero for 246. And zero is a network address. Again, not good.

treii28, 2019-03-08:
And I'm not understanding how your step 2) is working if you are responding with a different dns address for the urls. The ESP won't try to answer it if the incoming request isn't sent to it's AP address.

treii28, 2019-03-08:
nope - no helper pop-up when returning a different address for the googles. Same ole 'Internet may not be available' with no redirect to a landing page.

[31 more replies]

closed 2025-01-26 by idolpx