idolpx/mobile-rr #5: Error when compiling: invalid cast from type 'IPAddress' to type 'u8_t* {aka unsigned char*}'

Traste, 2016-08-03:
When trying to compile the project in PlatformIO (IDE 1.2.0 and CLI 2.11.2) - installed fresh on a Windows  10 Pro 64bit machine because of this project - and project downloaded from github today (after several failed attempts to compile and upload it using Arduino IDE) i receive errors while compiling:

In file included from C:\Users\morfo\.platformio\packages\framework-arduinoespressif\tools\sdk\include/user_interface.h:11:0,

from C:/Users\morfo\Documents/platformio/mobile-rr-master/src/mobile-rr.ino:42:
C:/Users\morfo\Documents/platformio/mobile-rr-master/src/mobile-rr.ino: In function 'void setup()':
C:\Users\morfo\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:220:43: error: invalid cast from type 'IPAddress' to type 'u8_t* {aka unsigned char*}'
#define ip4_addr1(ipaddr) (((u8_t*)(ipaddr))[0])
^
C:\Users\morfo\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:226:38: note: in expansion of macro 'ip4_addr1'
#define ip4_addr1_16(ipaddr) ((u16_t)ip4_addr1(ipaddr))
^
C:\Users\morfo\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:240:24: note: in expansion of macro 'ip4_addr1_16'
#define IP2STR(ipaddr) ip4_addr1_16(ipaddr), \
^
C:/Users\morfo\Documents/platformio/mobile-rr-master/src/mobile-rr.ino:329:3: note: in expansion of macro 'IP2STR'
) ;
^
C:\Users\morfo\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:221:43: error: invalid cast from type 'IPAddress' to type 'u8_t* {aka unsigned char*}'
#define ip4_addr2(ipaddr) (((u8_t*)(ipaddr))[1])
^
C:\Users\morfo\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:227:38: note: in expansion of macro 'ip4_addr2'
#define ip4_addr2_16(ipaddr) ((u16_t)ip4_addr2(ipaddr))
^
C:\Users\morfo\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:241:5: note: in expansion of macro 'ip4_addr2_16'
ip4_addr2_16(ipaddr), \
^
C:/Users\morfo\Documents/platformio/mobile-rr-master/src/mobile-rr.ino:329:3: note: in expansion of macro 'IP2STR'
) ;
^
C:\Users\morfo\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:222:43: error: invalid cast from type 'IPAddress' to type 'u8_t* {aka unsigned char*}'
#define ip4_addr3(ipaddr) (((u8_t*)(ipaddr))[2])
^
C:\Users\morfo\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:228:38: note: in expansion of macro 'ip4_addr3'
#define ip4_addr3_16(ipaddr) ((u16_t)ip4_addr3(ipaddr))
^
C:\Users\morfo\.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h:242:5: note: in expansion of macro 'ip4_addr3_16'
ip4_addr3_16(ipaddr), \
^
C:/Users\morfo\Documents/platformio/mobile-rr-master/src/mobile-rr.ino:329:3: note: in expansion of macro 'IP2STR'
) ;
^
...(several more repetitions of the above)
This is incidentally exactly the same error i gut stuck at in the Arduino IDE and the reason i decided to give PlatformIO a try.

Traste, 2016-08-03:
After some semi-random troubleshooting it seems like there has been some changes to lwip that has broken the code in your project. I tested replacing the 
.platformio\packages\framework-arduinoespressif\tools\sdk\lwip\include/lwip/ip_addr.h
file with an older version from 2015 i found somewhere, and that generated completely new errors from other lwip files called from ip_addr.h and that had not been replaced.
Comparing the ip_addr.h file that came with PlatformIO when i installed it, with the one from 2015, there were substantial changes and it seems to have been completely rewritten
attaching the two versions of ip_addr.h so you can compare with the one you have:
[lwip.zip](https://github.com/idolpx/mobile-rr/files/400256/lwip.zip)

idolpx, 2016-08-10:
Ok... got a solution.  Change the build flags to this.

build_flags = -ULWIP_OPEN_SRC,-Wl,-Tesp8266.flash.4m.ld

closed 2016-08-10 by idolpx