idolpx/mobile-rr #8: Ideas for Improvements

Noki, 2016-08-17:
I just started playing around with this and I have some ideas for improvements:

- Allow to set the channel for the AP via console
- Autodetect crowded channels and use best channel
- Allow to set the Rick-Roll message via console to send a special greeting
- Convert l.png to svg to reduce size and have a version that works on large monitors as well
- Add responsive CSS for large screens

idolpx, 2016-08-19:
Good ideas.  I will have to do some research on the first two.  I don't know how or if it's possible to set the channel.  It would be cool though.

Setting the message was something that I've thought about but wasn't sure if it would be worth having that message take up space in memory when running.  Although it could be written out to the SPIFFS file system and loaded from there dynamically.  

I did try converting l.png to svg before my first release but the file size came out bigger.  I used Inkscape.  I may have done it wrong though.

The index.htm file was originally designed using jQuery Mobile.  I ripped just what I needed to speed up the load time.  I'm sure making it responsive could be done without too much extra loading.  I'd be curious to see what you come up with.

Noki, 2016-08-19:
Setting the channel is as easy as WiFi.softAP(tssid, password, channel); An empty password can be used.

For the SVG there is a wonderful site (https://jakearchibald.github.io/svgomg/) to optimize those. I already played around with the file but I was not sure about the copyright of the file after I found it used on several other sites.

The index file is responsive, but not really good looking on large screens like notebooks. Thats what I ment.

idolpx, 2016-08-19:
Oh cool!  Code to come shortly.  :)

Yes... I found the image with a google search.  I figured that since I was not using it for a commercial purpose, it would be ok.

idolpx, 2016-08-19:
I'm sure this could be optimized but it should do the trick.

```
// WiFi Encryption Types
String encryptionTypes(int which) {
    switch (which) {
        case ENC_TYPE_WEP: return "WEP\t"; break;
        case ENC_TYPE_TKIP: return "WPA/TKIP"; break;
        case ENC_TYPE_CCMP: return "WPA2/CCMP"; break;
        case ENC_TYPE_NONE: return "None\t"; break;
        case ENC_TYPE_AUTO: return "Auto\t"; break;
        default: return "Unknown"; break;
    }
}

void setupAP()
{
    char no_pass[] = "\0";

    int chan_selected;
    int channels[14];
    std::fill_n(channels, 14, 0);

    WiFi.mode( WIFI_STA );
    Serial.println( "Scanning WiFi Networks" );
    int networks = WiFi.scanNetworks();
    Serial.printf( "%d Networks Found\n", networks );
    Serial.printf( "RSSI\tCHANNEL\tENCRYPTION\tBSSID\t\t\tNAME\n" );

    for (int network = 0; network < networks; network++) {
        String ssid_scan;
        int32_t rssi_scan;
        uint8_t sec_scan;
        uint8_t* BSSID_scan;
        int32_t chan_scan;
        bool hidden_scan;
        WiFi.getNetworkInfo(network, ssid_scan, sec_scan, rssi_scan, BSSID_scan, chan_scan, hidden_scan);

        Serial.printf( "%d\t%d\t%s\t%02X:%02X:%02X:%02X:%02X:%02X\t%s\n",
                        rssi_scan,
                        chan_scan,
                        encryptionTypes( sec_scan ).c_str(),
                        MAC2STR( BSSID_scan ),
                        ssid_scan.c_str()
                    );

        channels[ chan_scan ]++;
    }

    // Find least used channel
    int lowest_count = 10000;
    for (int channel = 1; channel<= 14; channel++){
        if ( channels[channel] < lowest_count )
        {
            lowest_count = channels[channel];
            chan_selected = channel;
        }
        //Serial.printf( "Channel %d = %d\n", channel, channels[channel]);
    }
    Serial.printf ( "Channel %d Selected!\n", chan_selected );

    WiFi.mode ( WIFI_AP );
    WiFi.softAPConfig ( ip, ip, IPAddress ( 255, 255, 255, 0 ) );
    WiFi.softAP ( ssid, no_pass, chan_selected );
}

RudyFiero, 2016-08-20:
"I figured that since I was not using it for a commercial purpose, it would be ok."
I think you would be safest if you state that it is for Educational purposes. And that has been your purpose.
http://copyrightuser.org/topics/education/

idolpx, 2016-08-20:
Good point.  And it is true that this is for educational purposes.

I've added the ability to change the Rick Roll Message.
You can also now set the Username and Password from the console too.

All of these changes are on the Enhanced Console branch for now.
I will merge them into master soon.

napkin2, 2016-08-20:
SVG:

http://pastebin.com/en5Krr8v - Raw
http://pastebin.com/K1btZjjq - Optimized

idolpx, 2016-08-20:
Oh awesome.  What did you use to convert the image?  Or did you find the vector image somewhere on the net?

napkin2, 2016-08-20:
I used http://vectormagic.com/home to convert and afterwards Inkscape to rework everything and https://jakearchibald.github.io/svgomg/ for compression. The portal loads much faster with inline SVG.

Noki, 2016-08-22:
Hi! 

This all looks great! Looks like there is not much left to contribute.

Here are some thoughts:

Wifi channels in 2.4 GHz overlap (https://en.wikipedia.org/wiki/List_of_WLAN_channels) and therefore it might be a good idea to just choose between channel 1, 6 and 11. Channel 14 is forbidden in most countries, so it is not really an option to use it or a lot of people would have to change the code to stay within the law. If you totally want to overengeneer it you could count stations on nearby channels with a percentage and rescan for the best channel from time to time (especially useful if you have it in your pocket and walk around at a conference).

The SVG really helps a lot, however I would have used it an inline SVG to reduce the number of HTTP-Requests and I even thought about replacing the d.gif with an animated SVG or CSS-Loader as well. I prepared one using http://loading.io/ and a CSS minifier.

```html
<style type='text/css'>
width:100%; @-webkit-keyframes uil-bar{0-ms-transform:scale(2);-moz-transform:scale(2);-webkit-transform:scale(2);-o-transform:scale(2);transform:scale(2)}100%,90%{-ms-transform:scale(1);-moz-transform:scale(1);-webkit-transform:scale(1);-o-transform:scale(1);transform:scale(1)}@-ms-keyframes uil-bar{0%{-ms-transform:scale(2);-moz-transform:scale(2);-webkit-transform:scale(2);-o-transform:scale(2);transform:scale(2)}100%,90%{-ms-transform:scale(1);-moz-transform:scale(1);-webkit-transform:scale(1);-o-transform:scale(1);transform:scale(1)}}@-moz-keyframes uil-bar{0%{-ms-transform:scale(2);-moz-transform:scale(2);-webkit-transform:scale(2);-o-transform:scale(2);transform:scale(2)}100%,90%{-ms-transform:scale(1);-moz-transform:scale(1);-webkit-transform:scale(1);-o-transform:scale(1);transform:scale(1)}}@-webkit-keyframes uil-bar{0%{-ms-transform:scale(2);-moz-transform:scale(2);-webkit-transform:scale(2);-o-transform:scale(2);transform:scale(2)}100%,90%{-ms-transform:scale(1);-moz-transform:scale(1);-webkit-transform:scale(1);-o-transform:scale(1);transform:scale(1)}}@-o-keyframes uil-bar{0%{-ms-transform:scale(2);-moz-transform:scale(2);-webkit-transform:scale(2);-o-transform:scale(2);transform:scale(2)}100%,90%{-ms-transform:scale(1);-moz-transform:scale(1);-webkit-transform:scale(1);-o-transform:scale(1);transform:scale(1)}}@keyframes uil-bar{0%{-ms-transform:scale(2);-moz-transform:scale(2);-webkit-transform:scale(2);-o-transform:scale(2);transform:scale(2)}100%,90%{-ms-transform:scale(1);-moz-transform:scale(1);-webkit-transform:scale(1);-o-transform:scale(1);transform:scale(1)}}.uil-bar-css{background:0 0;position:relative;width:200px;height:200px}.uil-bar-css>div{position:absolute;width:40px;height:120px;top:40px;left:20px;background:#8bc53f;opacity:.6;-ms-animation:uil-bar .65s cubic-bezier(.01,.73,.28,.93) infinite;-moz-animation:uil-bar .65s cubic-bezier(.01,.73,.28,.93) infinite;-webkit-animation:uil-bar .65s cubic-bezier(.01,.73,.28,.93) infinite;-o-animation:uil-bar .65s cubic-bezier(.01,.73,.28,.93) infinite;animation:uil-bar .65s cubic-bezier(.01,.73,.28,.93) infinite}.uil-bar-css>div:nth-of-type(2){left:80px;opacity:.8;-ms-animation-delay:60ms;-moz-animation-delay:60ms;-webkit-animation-delay:60ms;-o-animation-delay:60ms;animation-delay:60ms}.uil-bar-css>div:nth-of-type(3){left:140px;opacity:.9;-ms-animation-delay:.13s;-moz-animation-delay:.13s;-webkit-animation-delay:.13s;-o-animation-delay:.13s;animation-delay:.13s}
 </style> 
<div class='uil-bar-css' style='-webkit-transform:scale(0.13)'><div></div><div></div><div></div></div>
```

The code has more bytes as long as it is uncompressed but it can be compressed to be smaller than the gif. An SVG would be even smaller but Internet Explorer and Edge do not support animated SVGs.

boneskull, 2016-08-22:
>  Add responsive CSS for large screens

I have a [branch](/boneskull/mobile-rr/tree/html) where I'm working on some responsive stuff.  I'll send a PR if/when I'm happy with it--don't wait on me, though.  :wink:

idolpx, 2016-08-24:
Good point on the limiting the Max Channel to 11.  I thought about it but decided to go with full range figuring that it would be very unlikely that someone would notice get in trouble for it.  It is best to be cautious though.

How about including the two side channels in the count when determining least used channel?  Something like this should do the trick.

```
    // Find least used channel
    int lowest_count = 10000;
    for (int channel = 1; channel<= 11; channel++) {
        // Include side channels to account for signal overlap
        int current_count = 0;
        for ( int i = channel - 2; i <= (channel + 2); i++ ) {
            if (i > 0)
                current_count += channels[i];
        }
        if ( current_count < lowest_count )
        {
            lowest_count = current_count;
            chan_selected = channel;
        }
        //Serial.printf( "Channel %d = %d\n", channel, channels[channel]);
    }
```

I will do some tests with making the l.svg inline and your changes to make the d.gif into an animated SVG.  The more bytes saved the better. 😄

idolpx, 2016-08-24:
It would be cool to have PlatformIO gzip the appropriate data files when building the SPIFFS image.  I'm going to look into this and see what I can come up with.

I like the idea of having a page to explain what a Rick Roll is.  Maybe link the RR Message to that page.  Anyone want to throw something together for that?

Noki, 2016-08-24:
I thought about the code for the channel selection and modified it slightly:

```
// WiFi Encryption Types
String encryptionTypes(int which) {
    switch (which) {
        case ENC_TYPE_WEP: return "WEP\t"; break;
        case ENC_TYPE_TKIP: return "WPA/TKIP"; break;
        case ENC_TYPE_CCMP: return "WPA2/CCMP"; break;
        case ENC_TYPE_NONE: return "None\t"; break;
        case ENC_TYPE_AUTO: return "Auto\t"; break;
        default: return "Unknown"; break;
    }
}

void setupAP()
{
    char no_pass[] = "\0";

    int chan_selected;
    int channels[14];
    std::fill_n(channels, 14, 0);

    WiFi.mode( WIFI_STA );
    Serial.println( "Scanning WiFi Networks" );
    int networks = WiFi.scanNetwork();
    Serial.printf( "%d Networks Found\n", networks );
    Serial.printf( "RSSI\tCHANNEL\tENCRYPTION\tBSSID\t\t\tNAME\n" );

    for (int network = 0; network < networks; network++) {
        String ssid_scan;
        int32_t rssi_scan;
        uint8_t sec_scan;
        uint8_t* BSSID_scan;
        int32_t chan_scan;
        bool hidden_scan;
        WiFi.getNetworkInfo(network, ssid_scan, sec_scan, rssi_scan, BSSID_scan, chan_scan, hidden_scan);

        Serial.printf( "%d\t%d\t%s\t%02X:%02X:%02X:%02X:%02X:%02X\t%s\n",
                        rssi_scan,
                        chan_scan,
                        encryptionTypes( sec_scan ).c_str(),
                        MAC2STR( BSSID_scan ),
                        ssid_scan.c_str()
                    );

        channels[ chan_scan ]++;
    }

	// Find least used channel
	// Note: Good design practice is to use channels with no overlap, like most devices do
	// http://www.espressif.com/sites/default/files/esp8266_wi-fi_channel_selection_guidelines.pdf
    int lowest_count = 10000;
    for (int channel = 1; channel<= 11; channel++) {
        // Include side channels to account for signal overlap
        int current_count = 0;
        for ( int i = channel - 2; i <= (channel + 2); i++ ) {
            if (i > 0)
                current_count += channels[i];
        }
				// Select one of the channels with no overlap
        if ( ( current_count < lowest_count ) && ( channel == 1 || channel == 6 || channel == 11 ) )
        {
            lowest_count = current_count;
            chan_selected = channel;
        }
        Serial.printf( "Channel %d = %d\n", channel, current_count);
    }
		Serial.printf( "Best Channel = %d\n", chan_selected);

    WiFi.mode ( WIFI_AP );
    WiFi.softAPConfig ( ip, ip, IPAddress ( 255, 255, 255, 0 ) );
    WiFi.softAP ( ssid, no_pass, chan_selected );
}
```

To select only between the channels 1, 6 and 11 is the smart choise and it is exactly what most of the other stations do:

[image]

While testing the code I recognized, that the scan on the ESP8266 does run very fast and misses a lot of stations. Therefore this code needs some more work. I suggest we run 10 scans at startup, extract all the unique BSSIDs and perform the channel selection based on that list.

idolpx, 2016-08-24:
Check the "Enhanced Console" branch.  I just added a lot of changes.  :)

Noki, 2016-08-24:
Looks like I need to figure out how to get this flashed:

```
Check program size...
text       data     bss     dec     hex filename

320428     8940   31944  361312   58360 .pioenvs/d1_mini/firmware.elf
MethodWrapper(["upload"], [".pioenvs/d1_mini/firmware.bin"])
Looking for upload port/disk...
Manually specified: 10.10.10.1
"/home/tobias/.atom/packages/platformio-ide/penv/bin/python2.7" "/home/tobias/.platformio/packages/framework-arduinoespressif/tools/espota.py" --debug --progress -i 10.10.10.1 --auth=password -f .pioenvs/d1_mini/firmware.bin
23:08:35 [DEBUG]: Options: {'esp_ip': '10.10.10.1', 'host_port': 33506, 'image': '.pioenvs/d1_mini/firmware.bin', 'host_ip': '0.0.0.0', 'auth': 'password', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
23:08:35 [INFO]: Starting on 0.0.0.0:33506
23:08:35 [INFO]: Upload size: 333520
23:08:35 [INFO]: Sending invitation to: 10.10.10.1
23:08:45 [ERROR]: No Answer
scons: *** [upload] Error 1
========================= [ ERROR ] Took 13.75 seconds =========================
```

Noki, 2016-08-24:
ahh, disable over the air... ;-)

Noki, 2016-08-24:
When I try to build I get:

`Warning! Ignore unknown `custom_option` option from `[env:]` section`

and the build fails with:

```
...
Processing .pioenvs/d1_mini/firmware.elf
.pioenvs/d1_mini/lib/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_port_read':
WiFiClientSecure.cpp:(.text.ax_port_read+0x14): multiple definition of `ax_port_read'
.pioenvs/d1_mini/lib/libESPAsyncTCP_ID305.a(lwipr_compat.o):lwipr_compat.c:(.text.ax_port_read+0x14): first defined here
.pioenvs/d1_mini/lib/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_port_write':
WiFiClientSecure.cpp:(.text.ax_port_write+0x1c): multiple definition of `ax_port_write'
.pioenvs/d1_mini/lib/libESPAsyncTCP_ID305.a(lwipr_compat.o):lwipr_compat.c:(.text.ax_port_write+0xc): first defined here
.pioenvs/d1_mini/lib/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_get_file':
WiFiClientSecure.cpp:(.text.ax_get_file+0x0): multiple definition of `ax_get_file'
.pioenvs/d1_mini/lib/libESPAsyncTCP_ID305.a(lwipr_compat.o):lwipr_compat.c:(.text.ax_get_file+0x0): first defined here
.pioenvs/d1_mini/lib/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_wdt_feed':
WiFiClientSecure.cpp:(.text.ax_wdt_feed+0x8): multiple definition of `ax_wdt_feed'
.pioenvs/d1_mini/lib/libESPAsyncTCP_ID305.a(lwipr_compat.o):lwipr_compat.c:(.text.ax_wdt_feed+0x0): first defined here
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/d1_mini/firmware.elf] Error 1
========================== [ERROR] Took 5.17 seconds ==========================
```

I think I have to check this out with some more time... :-(

idolpx, 2016-08-24:
I just merged everything to master.  I did forget to comment out the OTA update lines.  I'm connecting to it via wifi while I'm working on it.  The OTA flash is pretty fast.  I wasn't able to get the 921600 baud to work with my D1 mini for some reason.

idolpx, 2016-08-25:
Did you try "Clean" before building again?  That may resolve the issue.

Noki, 2016-08-25:
I did try "Clean" and even deleted everything and reinstalled from scratch and still the same error. Same on my office computer. ;-)

idolpx, 2016-08-25:
I updated ESPAsyncTCP and ESPAsyncWebServer and started getting the same errors.
I posted my solution in the other issue thread.  Thanks for the troubleshooting on that.

closed 2016-11-07 by idolpx