idolpx/mobile-rr #12: multiple definition of `ax_port_write'

martinlbb, 2016-08-25:
When trying to compile this project following HOWTO, I have multiple errors:

```
WiFiClientSecure.cpp:(.text.ax_port_write+0x1c): multiple definition of `ax_port_write'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(lwipr_compat.o):lwipr_compat.c:(.text.ax_port_write+0xc): first defined here
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_get_file':
WiFiClientSecure.cpp:(.text.ax_get_file+0x0): multiple definition of `ax_get_file'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(lwipr_compat.o):lwipr_compat.c:(.text.ax_get_file+0x0): first defined here
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_wdt_feed':
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_wdt_feed':
WiFiClientSecure.cpp:(.text.ax_wdt_feed+0x8): multiple definition of `ax_wdt_feed'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(lwipr_compat.o):lwipr_compat.c:(.text.ax_wdt_feed+0x0): first defined here
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/d1_mini/firmware.elf] Error 1
```

I have latest Platform IDE installed, running IDE v1.4.0, and CLI 2.11.2
Platform espressif is installed via CLI.

Any idea?? :)

martinlbb, 2016-08-25:
Seems there is a multiple declaration on these files:

- libESP8266WiFi.a
- libESPAsyncTCP_ID305.a

```
.pioenvs/d1_mini/libESP8266WiFi.a(WiFiClientSecure.o): In function `ax_port_read':
WiFiClientSecure.cpp:(.text.ax_port_read+0x14): multiple definition of `ax_port_read'
.pioenvs/d1_mini/libESPAsyncTCP_ID305.a(lwipr_compat.o):lwipr_compat.c:(.text.ax_port_read+0x14): first defined here
```

martinlbb, 2016-08-25:
This may be related:
https://github.com/platformio/platformio/issues/659

martinlbb, 2016-08-25:
Previous idea was not related. 
In fact, I add to go to my .platformio/src/ library and manually rename offending functions in ESPAsyncTCP_ID305/src/lwipr_compat.c 

Offending functions are:
```
ax_port_read
ax_port_write
ax_get_file
ax_wdt_feed
```

It is also possible to use an ifndef call.

Noki, 2016-08-25:
The latest version of ESPAsyncTCP is broken. You can get it to compile by going back to a working version. Go to github and fetch this one e.g. by downloading the zip. Replace the files in the library folder.

https://github.com/me-no-dev/ESPAsyncTCP/tree/5987225cf66930b307a58f3ede3ba08830ee1c7b

However when I flash that firmware it hangs in a bootloop. Maybe you have more luck.

martinlbb, 2016-08-25:
Thanks @Noki for your imput. Well, I didn't changed ESPAsyncTCP, I just changed ESPAsyncTCP_ID305 (another different lib).

After compiling, uploading firmware, uploading SPIFFS (see note about it on web README, there is a picture showing it), it works.

I did it with an ESP-12-Q (small stamp version w/o usb serial). If I powered it with a serial-USB module, it does not work (not enough juice). If I power up with regular 3.3V, everything okay

idolpx, 2016-08-25:
Ok...  try this.  Open the terminal in PlatformIO and enter the following commands.

```
> pio lib uninstall 305 306
> pio lib install 305 --version=5987225cf6
> pio lib install 306 --version=dab3077f2b
```

That will revert those libraries to the previous versions that will actually compile.

closed 2016-08-25 by idolpx

Noki, 2016-08-26:
I found out what caused the bootloop. The eeprom could not be read and did not get initialized automatically. I got it working again by calling:

```
eepromInitialize();
eepromSave();
```

martinlbb, 2016-08-26:
Thanks all :)