idolpx/mobile-rr #20: error: '_ws_client' was not declared in this scope

marqueeeeeee, 2017-01-25:
Error compiling using PlatformIO, is there something missing from the libraries?

error: '_ws_client' was not declared in this scope
Compiling .pioenvs\d1_mini\FrameworkArduino\debug.o
Compiling .pioenvs\d1_mini\FrameworkArduino\heap.o
*** [.pioenvs\d1_mini\src\mobile-rr.ino.o] Error 1

idolpx, 2017-01-26:
Did you try doing a clean and then rebuilding?  
I have not gotten that error before.

marqueeeeeee, 2017-01-26:
yes, I did. still can't find a solution. 

steps that i did:
1. cloned the solution
2. installed platformIO and its dependencies (Windows)
3. Initialized the project - successful
4. selected ESP board - successful 
5. Rebuilt C/C++ - Successful
6. Update platforms, package and libraries - Successful
7.  Clean - Successful 
8. Build - Error

`int i, duration;
^
In file included from C:\users\--\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include/user_interface.h:11:0,
from D:/esp/project/mobile-rr/src/mobile-rr.ino:97:
D:/esp/project/mobile-rr/src/mobile-rr.ino: In function 'void setup()':
C:\users\--\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:220:43: error: invalid cast from type 'IPAddress' to type 'u
8_t* {aka unsigned char*}'`

idolpx, 2017-01-26:
See if it will build with the "build_flags" that are commented out in the platformio.ini.
It's the line just above the current "build_flags" setting.  Just comment the current line and uncomment the other.

marqueeeeeee, 2017-01-27:
Still having that error :( here's the full build logs. still finding another way, can i port this project to arduinoIDE? 

`^
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:226:38: note: in expansion of macro 'ip4_addr1'
#define ip4_addr1_16(ipaddr) ((u16_t)ip4_addr1(ipaddr))
^
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:240:24: note: in expansion of macro 'ip4_addr1_16'
#define IP2STR(ipaddr) ip4_addr1_16(ipaddr), \
^

D:/esp/project/mobile-rr/src/mobile-rr.ino:2651:28: note: in expansion of macro 'IP2STR'
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:221:43: error: invalid cast from type 'IPAddress' to type 'u8_t* {aka unsigned char*}'
#define ip4_addr2(ipaddr) (((u8_t*)(ipaddr))[1])
^
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:227:38: note: in expansion of macro 'ip4_addr2'
#define ip4_addr2_16(ipaddr) ((u16_t)ip4_addr2(ipaddr))
^
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:241:5: note: in expansion of macro 'ip4_addr2_16'
ip4_addr2_16(ipaddr), \
^
D:/esp/project/mobile-rr/src/mobile-rr.ino:2651:28: note: in expansion of macro 'IP2STR'
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:222:43: error: invalid cast from type 'IPAddress' to type 'u8_t* {aka unsigned char*}'
#define ip4_addr3(ipaddr) (((u8_t*)(ipaddr))[2])
^
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:228:38: note: in expansion of macro 'ip4_addr3'
#define ip4_addr3_16(ipaddr) ((u16_t)ip4_addr3(ipaddr))
^
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:242:5: note: in expansion of macro 'ip4_addr3_16'
ip4_addr3_16(ipaddr), \
^
D:/esp/project/mobile-rr/src/mobile-rr.ino:2651:28: note: in expansion of macro 'IP2STR'
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:223:43: error: invalid cast from type 'IPAddress' to type 'u8_t* {aka unsigned char*}'
#define ip4_addr4(ipaddr) (((u8_t*)(ipaddr))[3])
^
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:229:38: note: in expansion of macro 'ip4_addr4'
#define ip4_addr4_16(ipaddr) ((u16_t)ip4_addr4(ipaddr))
^
C:\users\mark.sancho\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include/lwip/ip_addr.h:243:5: note: in expansion of macro 'ip4_addr4_16'
ip4_addr4_16(ipaddr)
^
D:/esp/project/mobile-rr/src/mobile-rr.ino:2651:28: note: in expansion of macro 'IP2STR'
Compiling .pioenvs\d1_mini\FrameworkArduino\Stream.o
*** [.pioenvs\d1_mini\src\mobile-rr.ino.o] Error 1
 [ERROR] Took 6.19 seconds
 
PlatformIO: BuildFile 0Project 0No IssuesREADME.md1:1
CRLF6 deprecationsUTF-8GitHub Markdownmaster5 updates
`

idolpx, 2017-01-28:
What does your platformio.ini file look like?

closed 2017-03-21 by idolpx