idolpx/mobile-rr #21: SSID changes not processed

petorian, 2017-03-14:
When changing the SSID, the change is not applied.
It seems line 477 causes this behaviour:
`if ( config.channel != chan_selected ) // || config.ssid != ssid) )`

If I change this to `if (1==1)` then the code after that is applied and the access point is spawned with the new SSID,

If I uncomment the OR clause then I receive an error that the config.ssid and ssid are not of the same type. 
I do not have a lot of knowledge about changing this type, but will dive into this. Any help of course is appreciated.

idolpx, 2017-03-21:
When I was working on the issue with changing the channel I commented that out and meant to come back and fix it.  Sorry about that.  

Check for my update.  That should do the trick.  ;)
I haven't tested it yet.  My main system is down at the moment.

petorian, 2017-03-22:
Thanks, i'll try the update ! 👍🏼

Dennis

idolpx, 2017-03-27:
Ok...  finally had some time to test things out.  
Pull these changes and you should be good.

I fixed the IP issue too that you told me about.
The default 10.10.10.1 IP should work now.

petorian, 2017-03-29:
Great,  I really appreciate it! Thanks for fixing it.