idolpx/mobile-rr #28: Unable to build -> '_ws_client' does not name a type

Tobse101, 2018-03-15:
Hello,
today I tried to compile and upload the code to my D1 mini but it wont work.

Here is my platformio.ini
https://hastebin.com/eqilubojez.makefile

And my console output with exact message:
https://hastebin.com/hesoyefumi.tex

Please share some knowledge 👍 
Thanks

krisajacobs, 2018-03-17:
I'm having the exact same problem.

idolpx, 2018-03-20:
I just did a "pio update" and "pio lib update" and everything compiles fine.
Have you made changes to the code?  Can you post your copy of "mobile-rr.ino"?

krisajacobs, 2018-03-20:
Just did the updates myself, got all Up to Date back on everything. Still get this at compile time:

D:\_PlatformIO\mobile-rr/data/
D:\_PlatformIO\mobile-rr/www/
"gz|css|js"
Compiling .pioenvs\d1_mini\src\mobile-rr.ino.cpp.o
Archiving .pioenvs\d1_mini\lib9f9\libArduinoJson_ID64.a
Compiling .pioenvs\d1_mini\libf74\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pioenvs\d1_mini\libf74\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Compiling .pioenvs\d1_mini\libf74\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs\d1_mini\libf74\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
Compiling .pioenvs\d1_mini\libf74\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
Compiling .pioenvs\d1_mini\libf74\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pioenvs\d1_mini\libf74\ESP8266WiFi\WiFiClient.cpp.o
D:/_PlatformIO/mobile-rr/src/mobile-rr.ino:189:9: error: expected unqualified-id before string constant
}
^
D:/_PlatformIO/mobile-rr/src/mobile-rr.ino:247:3: error: '_ws_client' does not name a type
{
^
D:/_PlatformIO/mobile-rr/src/mobile-rr.ino:293:1: error: '_ws_client' does not name a type
}
^
D:/_PlatformIO/mobile-rr/src/mobile-rr.ino: In function 'void onEvent(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, uint8_t*, size_t)':
D:/_PlatformIO/mobile-rr/src/mobile-rr.ino:2351:18: error: 'ws_client' was not declared in this scope
D:/_PlatformIO/mobile-rr/src/mobile-rr.ino:2399:18: error: 'ws_client' was not declared in this scope
D:/_PlatformIO/mobile-rr/src/mobile-rr.ino:2503:18: error: 'ws_client' was not declared in this scope
D:/_PlatformIO/mobile-rr/src/mobile-rr.ino: In function 'void execCommand(AsyncWebSocketClient*, char*)':
D:/_PlatformIO/mobile-rr/src/mobile-rr.ino:2589:18: error: 'ws_client' was not declared in this scope
*** [.pioenvs\d1_mini\src\mobile-rr.ino.cpp.o] Error 1
==================================================================== [ERROR] Took 5.43 seconds ====================================================================
The terminal process terminated with exit code: 1

krisajacobs, 2018-03-20:
I have made no modifications - I just did a fresh clone from what you've got here.

Thanks!

Tobse101, 2018-03-21:
A fresh re-download and import in pio solved the problem. Cant explain.. Thanks!

closed 2018-03-21 by Tobse101

krisajacobs, 2018-03-21:
I just did the same thing - same error for me unfortunately.