idolpx/mobile-rr #29: Can't log in to console

printersofa, 2018-05-25:
Hi

I set a new username and password for the console. I set the username as test and the password as test. For some reason, I can't login to the console with these credentials. I double checked I typed them correctly and have checked capitals, spelling, etc.

I have also attempted to reflash the firmware on the device - both parts. The SSID stays the same and I still can't log in.

Any advice on how to reflash it properly?

idolpx, 2018-06-24:
You could call "eepromInitialize()" to reset the eeprom settings or you could disable the part of the code that checks the username/password.  After that you can change them and then re-enable the code and reflash and you should be good.  :)

closed 2018-06-24 by idolpx

printersofa, 2018-06-29:
@idolpx Any ideas on how I could call eepromInitialize()?

idolpx, 2018-07-10:
In setup change

setupEEPROM();

to

eepromInitialize();

Then flash the ESP and boot it up.
Then shut it down and change it back and reflash and you should be good.