idolpx/mobile-rr #31: Platformio stuck on building

29G, 2019-03-06:
I tried platformio installed in Atom on one Windows computer, and with platformio installed in VSCode on another Windows computer.
When i try to build it always gets stuck on:
`Processing d1_mini (platform: espressif8266; framework: arduino; board: d1_mini)`
It does nothing at all

DanielL99, 2019-03-07:
Open a terminal and try to uninstall ArduinoJson with
`pio lib uninstall ArduinoJson`
And install ArduinoJson version 5 with
`pio lib install ArduinoJson@5.13.4`

For now this project needs an older Version of ArduinoJson

29G, 2019-03-07:
Thank you. Doing that significantly reduced the amount of errors but unfortunately there are still some and the code won't build.
[
[image]
](url)

treii28, 2019-03-13:
Yeah both ArduinoJson and ip_addr have been modified in the more recent esp8266. I ran into this also. Fixing the ArduinoJson lib in your system to a <6 version can avoid the ArduinoJson errors or you can convert the code (line by line) to the new format. (why they didn't rename the lib is beyond me when they change the fundamental way it works)

You can avoid the ip_addr error by changing the ip_addr reference in the code to ip4_addr (as opposed to ip6_addr). Again, why deprecated support wasn't included *smh*

Another thing that I might suggest is that if the functions are going to be referenced in the code before they are declared in the code (many of the setup* functions appear to have this issue) either re-order the code so the setup functions come first or declare them at the top somewhere. e.g.:

```
//
//******************************************************************************************
// Forward declaration of methods                                                          *
//******************************************************************************************
int frequency ( char note );
int setupAP ( int chan_selected );
void setupEEPROM();
void setupSPIFFS();
void setupDNSServer();
void setupHTTPServer();
void setupOTAServer();
int scanWiFi ();
void readFile ( String file );
String getSystemInformation();
String getApplicationSettings();
void onTimer ();
void eepromLoad();
void eepromSave();
void eepromInitialize();
String getEEPROM();
bool disconnectStationByIP ( IPAddress station_ip );
bool disconnectStationByMAC ( uint8_t *station_mac );
void wifi_handle_event_cb ( System_Event_t *evt );
void onRequest ( AsyncWebServerRequest *request );
void onEvent ( AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void *arg, uint8_t *data, size_t len );
void execCommand ( AsyncWebSocketClient *client, char *msg );
void client_status ( AsyncWebSocketClient *client );

```

I had to add those for it to compile properly in platformio from a standard gcc [cross compile] call

treii28, 2019-03-13:
I might also strongly suggest that if you are going to use a custom DNSServer, refactor the libs to rename the DNSServer as this should save a lot of headaches over people that have other versions installed. e.g. change DNSServer to rrDNSServer and then modify all references to it in your code and in the library/header

29G, 2019-03-13:
Thank you, changing to ip4_addr fixed it and it successfully built.

closed 2019-03-13 by 29G