idolpx/mobile-rr #33: Error uploading file system

29G, 2019-03-13:
After finally getting the code to compile (https://github.com/idolpx/mobile-rr/issues/31) I uploaded to my NodeMCU.
I got "Too many redirects" error and realized that I forgot to upload the SPIFFS file system.
But when I tried to upload the file system, by clicking "PIO Upload File System image", I got this error:

`*** [.pioenvs\d1_mini\spiffs.bin] Error 148`

[image]

closed 2019-03-14 by 29G

idolpx, 2019-03-15:
Sorry so slow... Did you get it uploaded?  I was getting that error and had to adjust a build setting the platformio.ini.  That change was updated in the repo.

29G, 2019-03-15:
Yes with the latest commits it builds and uploads the SPIFFS with no errors, after downgrading arduinojson to version 5 instead of 6. Thank you