idolpx/mobile-rr #34: Can't build mobile-rr in Atom or VSCode with Platformio

bruhmoment1919, 2019-11-02:
Hi

Even after downgrading ArduinoJson, the project still refuses to built. These are the errors I have:

`> Executing task: C:\Users\User\.platformio\penv\Scripts\platformio.exe run <

Processing d1_mini (framework: arduino; platform: espressif8266; board: d1_mini)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.2.3 > WeMos D1 R2 and mini
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0), framework-arduinoespressif8266 2.20502.0 (2.5.2)
Converting mobile-rr.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 5.13.4
|-- <ESPAsyncTCP> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP Async WebServer> 1.2.3
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 5.13.4
|-- <Hash> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer>
|   |-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
|-- <Ticker> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
TypeError: a2b_base64() argument 1 must be string or buffer, not None:
  File "C:\Users\User\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 156:
    env.SConscript(item, exports="env")
  File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\User\Desktop\mobile-rr-master\www_proc.py", line 193:
    options = base64.b64decode(ARGUMENTS.get("CUSTOM_OPTION"))
  File "C:\Users\User\.platformio\python27\Lib\base64.py", line 75:
    return binascii.a2b_base64(s)
=============================================================== [FAILED] Took 4.61 seconds ===============================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.`

I have tried these both in Atom and VSCode but it does not work in both. Is anyone able to help me here?

Many thanks

bruhmoment1919, 2019-11-04:
Tried on two different PCs

bruhmoment1919, 2019-11-04:
I am now receiving syntax errors. I have tried to downgrade to python 2 but platformio core I not know how to make python 2

idolpx, 2019-11-17:
Ok... I updated the code to work with the latest ArduinoJson and latest PlatformIO.

Try it now and let me know how it does for ya.  :)

bruhmoment1919, 2019-11-18:
Hi
Still no change unfortunately, I originally got an error about a print on line 72 which I managed to fix by adding brackets around it.

Now, I get this error
```
  File "E:\Downloads\mbr\mobile-rr-master\www_proc.py", line 77
 

    pattern = re.compile(ur'<link.*href=[\'"](.*?)[\'"].*/>')
 
                           ^
 
SyntaxError: invalid syntax
 
========================== [FAILED] Took 3.40 seconds ==========================
```

I think this error has something to do with Python 2 and 3 since Platformio now seems to run Python 3.

Thanks

idolpx, 2019-11-18:
I just tested in Atom and VSCode on my macbook and it compiles fine on each.
I will test on a Windows 10 VM shortly but everything seems to be fine now.

Maybe try uninstalling PlatformIO and starting over.  Hope you get it going.  :)
I haven't touched it in a while but it is a fun little project.

closed 2019-11-18 by idolpx