idolpx/mobile-rr #36: 'class DNSServer' has no member named 'onquery'

kodda74, 2020-08-11:
Whenever I try to upload I get the error 'class DNSServer' has no member named 'onquery'. Any suggestions?

idolpx, 2020-09-15:
Sorry so slow to reply.  It's been a while since I checked the code.  Did you get this issue figured out?

kodda74, 2020-09-15:
No. I tried to fix it for like two weeks but I never got it fixed. Do you know what might he causing the issue? If you have a precompiled version of the code that would be awesome too

idolpx, 2020-09-16:
I'm not having issues compiling. What OS are you using? Are you using PlatformIO in Atom or VSCode?

kodda74, 2020-09-16:
I got it fixed this morning.  It was an issue with my PlatformIO build.

closed 2020-09-16 by kodda74

idolpx, 2020-09-16:
Oh awesome!  :D That's good news!  Let me know what kind of fun you have with it. :)