idolpx/mobile-rr #37: serial.serialutil.SerialException: [Errno 2] could not open port /dev/cu.wchusbserial14440: [Errno 2] No such file or directory: '/dev/cu.wchusbserial14440'

zLeki, 2022-04-08:
New to all of this. Had some fun experimenting with my new ESP8266 until I stumbled upon this. I decided to try it but now im having issues. If anyone could help that would be nice

zLeki, 2022-04-08:
Additional note- It builds fine just does not upload correctly.

closed 2022-04-08 by zLeki

idolpx, 2022-04-12:
Cool... check the settings in platformio.ini and make sure the ports are set correctly for your system.  :)

zLeki, 2022-04-12:
> Cool... check the settings in platformio.ini and make sure the ports are set correctly for your system.  :)

Thanks yeah I figured it out. Stupid mistake from me

idolpx, 2022-04-15:
Oh good...  let me know if you have any questions or run into any probs with it.  :)
I'm happy to help.  You can get my attention faster on discord.  https://discord.gg/FwJUe8kQpS
I'm in there most of the time.