idolpx/mobile-rr #38: Can we still make this work?

testataaaa, 2022-04-29:
As a noob I saw this project in a recent edition of Make Magazine (April 1st pranks) and through it looked well documented and straightforward.

Unfortunately I can't get this to work according to the instructions, is this still functional? If not, is there a way to do this via Arduino IDE rather than platformio?

idolpx, 2022-05-02:
Oh awesome!  Yes!  Message me on Discord.  I'd be happy to help you get it working.
https://discord.gg/FwJUe8kQpS

idolpx, 2022-05-03:
Hey! What issue of the magazine was it in?  I'd like to check it out.  I didn't even know about that.

testataaaa, 2022-05-05:
Thanks for your reply, I'll check out the Discord. As reference for people stumbling upon this, it was in Make Magazine edition that came out around April, and it listed the project as a good April fool's prank.

closed 2022-05-05 by idolpx