idolpx/mobile-rr #41: Error Compiling

abenaki47, 2023-05-09:
Using VS and PIO, was able to compile using a PIO sample project, but not the mobile-rr. I get a lot of errors in the compile, probably something simple but I'm just a hardware guy and not a SW hacker. Here is a bit of the error message;
Compiling .pio\build\d1_mini_pro\src\mobile-rr.ino.cpp.o
In file included from .pio\libdeps\d1_mini_pro\ArduinoJson\src/ArduinoJson.hpp:42,
                 from .pio\libdeps\d1_mini_pro\ArduinoJson\src/ArduinoJson.h:9,
                 from C:/Users/rODNEY/Documents/RickRoll project/mobile-rr-master/mobile-rr-master/src/mobile-rr.ino:36:
.pio\libdeps\d1_mini_pro\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:84:19: error: template argument 1 is invalid

CrazyDoge128, 2023-10-25:
Hi there using vs code and PIO too same problem here I'm just some kid interested in esps and electronics in general. I think it has something to do with **convertFromJson** .
Tried to use 4yold libraries with no luck. Would anyone be able to help me or send me a [compiled bin file](https://community.platformio.org/t/how-to-make-bin-file-using-platformio-ide/8491)
Thanks in advance for helping me.
here is my output:
[terminal_output.txt](https://github.com/idolpx/mobile-rr/files/13170115/terminal_output.txt)

CrazyDoge128, 2023-10-25:
also **https://github.com/NopeNix/mobile-rr-2023-ger** 🤦

idolpx, 2023-11-03:
Hey... I just pushed some changes.  Not sure what's going on with your error but pull the current code and see if it will compile for you.

closed 2025-01-26 by idolpx